sandeepmistry/esp8266-Arduino #1: Working on Windows 7 64bits but nothing on Windows 8.1 64bits

luc-github, 2015-04-17:
Now system is enabled - I use it :smile: 
I am using arduino 1.6.3 on 2 my 2 systems I used the helloserver sample sketch

on Win7 all is Ok, compilation / upload 

on Win8.1 no error message but no compilation : 
```
Using library ESP8266WiFi in folder: C:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
Using library ESP8266WebServer in folder: C:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer 

C:\Users\luc\Documents\Arduino\hardware\esp8266com/esp8266/tools/windows/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -IC:\Users\luc\Documents\Arduino\hardware\esp8266com/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10603 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\variants\generic -IC:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src -IC:\Users\luc\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src C:\Users\luc\AppData\Local\Temp\build8670606031828182613.tmp\HelloServer.cpp -o C:\Users\luc\AppData\Local\Temp\build8670606031828182613.tmp\HelloServer.cpp.o 
Error compiling.
```

Anything I can do to debug/solve the issue ?

Thanks,

Makuna, 2015-04-18:
I can confirm this.  Win 8.1 64 bit, Error compiling, no extra information is displayed when turned on.

luc-github, 2015-04-18:
seems a missing DLL I compared my 2 systems and on one of them I have GIT for windows with a bunch of GNU tools/dll I have downloaded http://git-scm.com/download/win on not working system and it is fine now

luc-github, 2015-04-18:
So I close issue

Makuna, 2015-04-18:
I really think this issue should be left open, and that it should be solved so that the Average Arduino user doesn't have to worry about this setup detail.  
But I can confirm that the above steps above will get the needed libiconv-2.dll onto your system so that you can make sure its on the path (or copied to the root of the Arduino folder).

sandeepmistry, 2015-04-18:
@Makuna agreed, re-opening ... unfortunately I don't have access to a Windows 8.1 64-bit system.

@luc-github @Makuna  have you tried putting the dll somewhere in the ```esp8266-Arduino/esp8266com/esp8266/tools/windows``` folder?

@rogerclarkmelbourne do you have any other suggestions on this?

brendanmatkin, 2015-04-18:
I can confirm the problem. Re-installing Git for Windows didn't work for me. libiconv-2.dll is in the 1.6.1 Arduino zip folder but NOT the 1.6.3. I copied libiconv-2.dll from 1.6.1 into my Arduino root folder (I have 1.6.3 installed via the installer) and now it is working.

luc-github, 2015-04-18:
I removed the dll from the path and tested it is not compiling
it is working if I copy the dll in 2 locations the 2 are mandatory if one is missing it still fail:
esp8266com\esp8266\tools\windows\xtensa-lx106-elf\libexec\gcc\xtensa-lx106-elf\4.8.2
esp8266com\esp8266\tools\windows\xtensa-lx106-elf\bin

so putting dll in these locations works

Hope it helps

rogerclarkmelbourne, 2015-04-19:
@luc-github 

I already tried that. 

There are some references in the docs that mention that putting the dll in libexec may work, but it didnt work for me. I tried putting the DLL in all folders that contained dll's or exes and it didnt help

The issue is because the compiler is compiled under Windows MinGW and it assumes it will be running in a similar environment, in which it can access libiconv-2.dll in the path.

You may find if you add somewhere in the hardware/arduinoesp8266 folder containing the DLL to your Windows path, that it would fix it

But that probably a more complex solution than simply copying to the Arduino installation folder

Albeit, if putting libiconv-2 into the Arduino install folder doesn't work, adding the location to the Windows path would be worth a try

luc-github, 2015-04-19:
I have deleted all libiconv-2.dll and libiconv2.dll (just in case of) of my system, I have scanned my disk, no more on my system so I am confident none is in the path, and if any place in the path compilation will works anyway.

So added the dll in 
esp8266com\esp8266\tools\windows\xtensa-lx106-elf\libexec\gcc\xtensa-lx106-elf\4.8.2
and
esp8266com\esp8266\tools\windows\xtensa-lx106-elf\bin

it works - I cannot comment on your system - I just say if not in path put the dll in these both locations solve my compilation problem. on Win7 and Win8.1

rogerclarkmelbourne, 2015-04-19:
@luc-github 

OK. So does it need to be on both places for it to work?

I trued in libexec and also in bin, but not in both folders at the same time.
It sounds like gcc is calling sub processes, each of which uses different path locations.

If you can get Sandeep to update his repo it would be the best outcome (until Ivan an make a release which does this)

Of if Sandeep is not available, I can update my repo to basically be the same as Sandeeps but with the fixes.

Or you could make your own repo !!

luc-github, 2015-04-19:
Yes as I wrote "copy the dll in 2 locations the 2 are mandatory if one is missing it still fail"
I think it would be good to validate this first before pushing to repo - as no one but me tested this

rogerclarkmelbourne, 2015-04-19:
OK

I guess the best thing is to download Sandeeps repo and apply it

sandeepmistry, 2015-04-19:
@luc-github @rogerclarkmelbourne if it works, please submit a pull request with the dll file in both locations - I'll gladly merge and make a new release/tag.

I don't have access to Window 8.1 64-bit PC, so don't know which dll to add, etc.

rogerclarkmelbourne, 2015-04-19:
I have Win 8.1 on X64 but don't normally use it for Arduino 

But I can install and test later

luc-github, 2015-04-19:
I would be happy to to the PR - I am going out now so it will be tonight
As I said I think it is better to validate first - but if no one can feedback - in worst case it won't hurt to add these 2 dll in module - it is up to you

rogerclarkmelbourne, 2015-04-19:
@luc-github 

Just tested again on my machine with 1.6.3 and I works without adding libiconv into the compiler

However this could be because I have the Due installed, because when I searched in %appdata%/Arduino15 , libiconv was in 2 places which looked like they were something to do with the Due (SAM) board.

I will need to test on a machine were I have not installed the Due

luc-github, 2015-04-19:
I removed my due files from board manager to be in worst case when testing as not every one download due package but I remembered I had issue even with due files

sandeepmistry, 2015-04-19:
@luc-github's pull request #4 has added the necessary dll in both places, I'll create a new tag shortly.

closed 2015-04-19 by sandeepmistry

sandeepmistry, 2015-04-19:
I've created a new 0.0.4 release which includes @luc-github changes.

commit b64079f5 (2015-03-30): Update to SDK 1.0.0

Fixes #1