sandeepmistry/esp8266-Arduino #2: Compiles on Linux but esptool not listed as a programmer

sej7278, 2015-04-17:
As mentioned in https://github.com/esp8266/Arduino/issues/14, with IDE 1.6.1 or 1.6.3 on Linux sketches compile and the hardware is listed, however esptool is not listed as a programmer.

I believe Mac users are having the same issue, although it seems to work for Windows apparently.

This is with a vanilla arduino.cc IDE not the esp8266-modified one.

luc-github, 2015-04-17:
In windows  esptool is not listed as a programmer but uploading/flashing is working

sej7278, 2015-04-17:
@luc-github how do you upload without setting a programmer - just hit the upload icon? Does that even work?

luc-github, 2015-04-17:
Yes - just hit upload button - it works if any programmer is selected or none

sej7278, 2015-04-17:
ah silly me! that should be documented in big **BOLD** letters!

anyway, now its trying to upload but just seems to timeout after a while:

```
Sketch uses 191,588 bytes (36%) of program storage space. Maximum is 524,288 bytes.
Uploading 35536 bytes from /tmp/build7803364969290628209.tmp/Blink.cpp_00000.bin to flash at 0x00000000
...................................
Uploading 156088 bytes from /tmp/build7803364969290628209.tmp/Blink.cpp_40000.bin to flash at 0x00040000
.........................................................................................................................................................

```

i've got a feeling this is why most of us dumped esptool and moved to calling esptool.py twice

luc-github, 2015-04-17:
when finished It show done uploading
[image]

sej7278, 2015-04-17:
yup, doesn't for me. esptool never worked for me using the regular sdk, freertos one or nodemcu, only esptool.py

i've never heard of esptool working on non-windows platforms.

luc-github, 2015-04-17:
May be it just do not give back the control when flash is finished - did you tried to see of your module is working as expected after upload look stuck ?

sej7278, 2015-04-17:
no, the module doesn't work as expected afterwards, no led's blink and there's gibberish on the serial console. flashing back to freertos works fine with esptool.py

luc-github, 2015-04-17:
if esptool.py is working you can change the command to use it no ?

in platform.txt just change cmd, path, params and pattern

tools.esptool.cmd=esptool
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path={runtime.hardware.path}/esp8266/tools/{runtime.os}

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=-vv
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp {serial.port} -ca 0x00000 -cf "{build.path}/{build.project_name}_00000.bin" -ca 0x40000 -cf "{build.path}/{build.project_name}_40000.bin"

sej7278, 2015-04-17:
yes that's what i'm working on now, but it needs to be called twice, so i'll have to do some dirty hacks lol

sej7278, 2015-04-17:
bizarrely the regular esptool seems to work again now, maybe esptool.py erasing the chip fixed something.

anyway, at least we have the option of using esptool.py

closed 2015-04-17 by sej7278

sandeepmistry, 2015-04-18:
@sej7278 great to hear you got things working! @luc-github thanks for helping out.

Please submit a pull request for any changes to the read me that would be helpful to others.

The next release of "main" ESP8266 repo will not need a programmer set as well, I've submitted a PR that has been merged already: https://github.com/esp8266/Arduino/pull/25

sej7278, 2015-04-18:
@sandeepmistry personally i don't see the point in the "main" repo, making a standalone platform core like you're doing is the way forward, maintaining a fork of the whole IDE is silly.

erniberni, 2015-05-10:
Hi,
in my case (Linux32bit) the esptool was also missing in the programmers menu and the sketch doesn't even compile and breaks with an esptool error (can't remember the detail). I checked the entries in the platform.txt where everything was fine. 
Then I found a programmers.txt file in the same directory that was empty. I copied this file from the Linux64Bit version to this place and the esptool was listed in the IDE under programmers.
Compile and upload now works fine.

sandeepmistry, 2015-05-10:
@erniberni esptool no longer needs to be selected as a programmer, this is why the programmers.txt is empty. You can just hit upload.

Are you sure you got things working on 32-bit Linux? My understanding is all the compilers are 64-bit.

erniberni, 2015-05-10:
@sandeepmistry
yes, I read before that the programmer.txt isn't needed, but the sketch did not compile and breaks with an esptool error. After adding the programmers.txt it worked. 
Now I removed the content of programmer.txt again, menu entry disappears, and now compiles without error. Don't know what happens. Anyway, it works.
Yes, I am working on 32bit Linux. I build everything from source

sandeepmistry, 2015-05-10:
@erniberni interesting, did you have the programmer selected before with another release? maybe the IDE got confused ...

Would you be able to share the 32-bit tool binaries you built? I can try to get them incorporated somehow then ...

erniberni, 2015-05-10:
I can't reproduce the error, maybe you're right.

The 32-bit code was build on Ubuntu 14.04.2 LTS, Linux version 3.13.0-52-generic (buildd@batsu) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #85-Ubuntu SMP Wed Apr 29 16:44:56 UTC 2015 .
I would share it for some time on Dropbox, so copy it from there and test it. Give me feedback, if it's working for you. Feel free to share it.
Here is the Dropbox link: https://www.dropbox.com/s/foiy2z6af5m8dzq/arduino-1.6.1-linux32.tar.xz?dl=0

tytower, 2015-05-10:
Yes that works for me too on 32 bit  but the IDE is 1.6.1 
Latest version of 1.6.4 is the problem for me

tytower, 2015-05-10:
Finally got my first chips programmed and working on the 64 bit machine and 1.6.4 IDE after moving libraries around .

Can also report that this 32 bit version with 1.6.1 also programs the chips fine 

Thank you Sandeep

commit cb799b67 (2015-03-30): fix platform file to support Arduino Eclipse Plugin

fixes #2