sandeepmistry/esp8266-Arduino #3: Missing lots of libraries

sej7278, 2015-04-17:
https://github.com/esp8266/Arduino/tree/esp8266/libraries

vs.

https://github.com/sandeepmistry/esp8266-Arduino/tree/master/esp8266com/esp8266/libraries

sej7278, 2015-04-17:
well the only esp8266-specific one that's missing is OneWire

sandeepmistry, 2015-04-18:
@sej7278 hmm, this wasn't on purpose, they must have got lost during a sync.

Should we re-open this? To at least add the OneWire lib? What about the others?

sej7278, 2015-04-18:
the OneWire library is esp8266-specific and doesn't come with the official IDE by default, i think the rest are part of the regular IDE, so don't need to be included in your platform core.

its only a few ifdefs different to the upstream library, so works fine with AVR boards still, so worth bundling it in your core certainly.

sandeepmistry, 2015-04-19:
Re-opening to track missing OneWire lib ...

sandeepmistry, 2015-04-19:
Address via https://github.com/sandeepmistry/esp8266-Arduino/commit/d669e77342cd69d03363e8160ce7329b52701a46

closed 2015-04-19 by sandeepmistry