sandeepmistry/esp8266-Arduino #5: Install steps for Windows have user copy into "user\documents" where "program files" is more appropriate

Makuna, 2015-04-19:
Your instructions state to copy to the "user\documents\arduino\hardware" folder.  This hardware folder doesn't exist here normally and having them in within the user folder can cause them to be sync'ed into OneDrive or other user backup software.

A more appropriate location is the "program files (x86)\ardunio\hardware" folder where the rest of the Arduino application bits are stored.

I have confirmed this on Win8 x64 and it should work for all Windows versions.

sandeepmistry, 2015-04-20:
@luc-github @rogerclarkmelbourne @brendanmatkin any thoughts on this?

rogerclarkmelbourne, 2015-04-20:
I'm not aware of the user Documents folder being synced 

I know that the Arduino IDE has an issue with this, because they now use %appdata% to store the ARM compiler and that is definately synced to the cloud for some people

I'm not sure if installing it in the program files still works. As of 1.6.2 all additional boards are now installed by the IDE into %appdata%/Arduino15/.......   

They did a bug fix in 1.6.3 and moved the AVR compiler back to the program files (I think) because the change broke a whole load of third party AVR boards

I've already raised the issue of using %appdata% for the compiler is in breach of Microsoft's guidelines for that folder, but this has basically been ignored by the dev team.
(I'm not the only one who reported this, and there are similar issues on Linux as they have put the SAM and Galileo files in the wrong place for them as well)

If you copy the files to the program files folder and the IDE detects the board, you'll still need to change the platform.txt file, as it will be referencing the documents/arduino/hardware/esp8266 folder structure for the compiler path

But its fairly easy to change the {} code

However, asking people to install stuff into program files, manually seems to be a bad idea

There must be some way to tell Windows not to sync the Documents folder to the cloud

Makuna, 2015-04-21:
I believe its worse to have application binaries put into the user/documents folder.

Having them in the documents folder, it restricts the use to the person who installed it while others who log into that computer will not have the support.  Having them installed in the program files gives the support to all users who log into the computer.

I copied the files into the program files directory (program files (x86)/arduino/hardware/) and it works without changing any paths, so I am unsure of the requirement/need to modify paths in the platform.txt file.

I have confirmed, that the "sync to cloud" can be set to exclude that folder, its just annoying that I have to do this.

rogerclarkmelbourne, 2015-04-21:
Did you remove the binaries from your documents folder before you did the test in the program files

I'm surprised that it found the compiler 

If you look in platform.txt
```
compiler.tools.path={runtime.hardware.path}/esp8266/tools/{runtime.os}/
compiler.path={compiler.tools.path}xtensa-lx106-elf/bin/
```

its not using the runtime.ide path, its using the runtime.hardware path (ie the users documents)

What Sandeep is doing is conforming to Arduino's own specification for how to do this

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

If you have an issue with conforming to the published Arduino spec, I'd recommend you post an issue on github.com/arduino/arduino  and subscribe to the developers mailing list (and post to that mailing list)
And take this up with the Arduino IDE team, rather than requesting Sandeep to change his repo to be non-conformant

luc-github, 2015-04-21:
I agree with @rogerclarkmelbourne is not great to use program files folder, especially when you are not admin of your PC or if you are updading the files often.
There is already a request to add function in board manager to upload zip that make sense : https://github.com/arduino/Arduino/issues/2949

In my opinion Documents is convenient place for current state of the module because subject to several update, and in future using board manager as zip or fully integrated will be the direction to go.

Also the problem of the sync is false problem 
[image] - it is always better to choose what you backup, not everything in my documents

 the not existence of hardware folder is also a weird reason IMHO

brendanmatkin, 2015-04-21:
In all previous versions of Arduino that I have used, user/documents/hardware is where other boards go. The nice thing about this is when you do a new install of Arduino or a modified version (e.g. Teensy), you don't have to reinstall your boards (sketches, libraries, etc.). The Arduino installer erases everything in the program files/Arduino. My windows 8.1 documents do not sync to anything, but I don't see any issue with having these be backed up. 

I am not familiar with 1.6.x best practices yet so I am only speaking from experience.

rogerclarkmelbourne, 2015-04-21:
@brendanmatkin 

Good point.

Re: 1.6.2 and newer,  practices are still in a state of flux. The IDE team added the Boards Manager which appears to a strategy to block Arduino.SLC boards being easily installable using the Arduno.llc IDE (e.g. Arduino vs Arduino battle in the courts)

I had issues with STM32 for Arduino issue not working and requested a method for it to be added via the Boards Manager, but as far as I'm aware the IDE team are not allowing third party boards to be installed via the Boards Manager at the moment

Adafruit have done a hack using a proxy server you have to run on your machine and some settings changes , in order to install their boards via the Boards Manager, and I think that one version of the arduino esp8266 can be installed via that method. But I"m not sure how it can work without libiconv2-dll being copied (but thats another story)

So basically as it stands 1.6.2 or newer is an equal mess, because %appdata% where they store all downloads and compilers etc etc etc, is definitely in the path that Windows 8 etc sync's  

What we have now is the lesser of the two evils

Makuna, 2015-04-21:
@rogerclarkmelbourne - yes, I removed from the documents\Arduino before testing and it works fine.

All good points listed, and I agree the Arduino IDE needs to be fixed (AppData is not for executables).  If they every turn this into a Windows Store App, they will have to fix it before it gets accepted.  

@brendanmatkin - Not requiring a delete of everything under program files/arduino before installing the next version but also not having a solution for boards is a big one.  This also is nothing new to application developers and there are "models" to follow where they remove everything but "extentions" folders.  Oh well.

I forgot how "convoluted" admin writes can be before Win8, its not so bad anymore but it does warn you about writing into this area but allows you do so with a default user account (I believe they are similar to the old Admin) as I never picked the level. 

Evil is relative ;-)

Glad we captured this, but I agree there is no clear cut "right" solution.

rogerclarkmelbourne, 2015-04-21:
I recall someone suggesting using  

C:\Program Files (x86)\Common Files\Arduino etc 

Also the IDE team mentioned something about not using appdata\roaming but to use \appdata\local

The latter may possibly be the best

There is another issue with putting keeping the cores (programming files) and the compiler together in some strange location
Because often these third party boards are in constant development, and its difficult to make code changes in program files (without running your code editor e.g. notepad++ as Administrator)

I know that perhaps most people won't be changing the source for the cores, but its not unheard of, 

Especially as the libraries are basically the same place and people make library changes more often

I believe the Arduino IDE team were planning to store the cores and the bins in different places, but possibly still not so that access to cores sources was easy (e.g not in a hidden or protected folder) :-(

Makuna, 2015-04-21:
Libraries can go either in \program files\Arduino\Libraries or \user\documents\Arduino\Libraries.  I put all extension libraries in \user\documents\Arduino\libraries.

rogerclarkmelbourne, 2015-04-21:
The problem with libraries is sometimes they need to be platform specific and don't work with the AVR or Due boards.

The IDE takes the ones in the user's documents folder in preference to libs that come with the hardware, which is often also and issue - but there is no simple way around this for 3rd party libs, However for things like Wire and SPI etc, having the platform specific libs in with the hardware normally makes a lot of sense

sandeepmistry, 2015-04-21:
Thanks for the feedback and info!

The consensus seems to be leave things as is.

@Makuna I think things should work if you copy to Program Files, the value of ```{runtime.hardware.path}``` is probably the location of ```platform.txt```. This add-on is relativity large compared to most, because it contains a toolchain as well.

closed 2015-04-21 by sandeepmistry