sandeepmistry/esp8266-Arduino #6: program space functionality is not abstracted (PgmSpace.h)

Makuna, 2015-04-22:
There are some libraries that try to reduce memory use and store string constants in program space.  While I am unsure if this is supported for Esp8266, and abstraction could be done to not use special handling while allowing libraries to be written generally between the platforms.

Makuna, 2015-04-23:
I added this snippit below to expose a few missed methods

#define strcasecmp_P(a, b)          strcasecmp((a), (b))

#define strncpy_P(dest, src, size) 		strncpy((dest), (src), (size))
#define strncat_P(dest, src, size) 		strncat((dest), (src), (size))
#define strncmp_P(a, b, size) 			strncmp((a), (b), (size))
#define strnlen_P(s, size) 				strnlen((const char *)(s), (size))
#define strncasecmp_P(a, b, size)       strcasecmp((a), (b), (size))

sandeepmistry, 2015-04-23:
@Makuna thanks for bringing this up, it's probably best to bring this up upstream (https://github.com/esp8266/Arduino) by either raising an issue or submitting a pull request.

I'd like to keep this fork in sync with the ESP8266 Arduino IDE, so if it gets addressed there, the changes can be pulled in here.

Thoughts?

Makuna, 2015-04-24:
I will get this done today and then close this.

Makuna, 2015-04-24:
https://github.com/esp8266/Arduino/pull/107

closed 2015-04-24 by Makuna

sandeepmistry, 2015-04-26:
@Makuna the fix for this has been merged in: https://github.com/sandeepmistry/esp8266-Arduino/commit/6a37b4461caa8bbff1ba473ec55f248cb5db3522