sandeepmistry/esp8266-Arduino #8: #include <string.h> doesn't allow access to c string functions

Makuna, 2015-04-23:
strlen is not accessible after including string.h.

luc-github, 2015-04-23:
I use 0.0.4 and strlen is working - I use #include <Arduino.h> but not string.h

Makuna, 2015-04-23:
To be specific, #include <Arduino.h> from the ino or a sketch header file works.  
From a library, #include <Arduino.h> does not work.  

I have a library that in the header it has a inline function that calls it and it won't compile.

Makuna, 2015-04-23:
It seems that strnlen is missing in some of the lower level code.

in libc_replacements.c, strlen is present but not strnlen
ets_strnlen is missing which means the rom doesn't support it.

Makuna, 2015-04-23:
I added the following to libc_replacements.c but this file doesn't seem to get compiled with the project.

size_t strnlen(const char *s, size_t len) {
    const char *cp; 
    for (cp = s; len != 0 && *cp != '\0'; cp++, len--); 
    return (size_t)(cp - s); 
}

sandeepmistry, 2015-04-23:
Once again, please raise an issue or submit a pull request upstream (https://github.com/esp8266/Arduino).

Makuna, 2015-04-24:
I will get this done today and then close this.

Makuna, 2015-04-24:
https://github.com/esp8266/Arduino/pull/108

closed 2015-04-24 by Makuna

sandeepmistry, 2015-04-26:
@Makuna the fix for this has been merged in: https://github.com/sandeepmistry/esp8266-Arduino/commit/eab9e873a6b6cc8ea589b1f1e6f15f65ec977697