sandeepmistry/esp8266-Arduino #9: std itoa is not found with arduino.h nor stdlib.h

Makuna, 2015-04-23:
including either of them doesn't work.
including stdlib_noniso.h does.

this makes it hard to write a simple library that can compile for AVR and ESP8266.

sandeepmistry, 2015-04-23:
Once again, please raise an issue or submit a pull request upstream (https://github.com/esp8266/Arduino).

Makuna, 2015-04-24:
I will get this done today and then close this.

Makuna, 2015-04-24:
https://github.com/esp8266/Arduino/issues/110

closed 2015-04-24 by Makuna

sandeepmistry, 2015-04-25:
@Makuna the fix for this is in master now: https://github.com/sandeepmistry/esp8266-Arduino/commit/6a7d94a2df6b89a7df9bf2c4cc792d34e94bdb25

Makuna, 2015-04-26:
I wish I could figure how to get the bits in place.  There are no instructions for building it and what I see it wants to replace Arduino 1.6.1 and I have other projects that require 1.6.3.

sandeepmistry, 2015-04-26:
@Makuna see: https://github.com/esp8266/Arduino/blob/esp8266/build/howto.txt

On OS X, I've renamed their ```Arduino.app``` to ```Arduino-ESP8266.app``` and have both running side by side. Not sure if the same can be done on windows.

I've enabled gitter, so let's follow up the discussion there: https://gitter.im/sandeepmistry/esp8266-Arduino

Makuna, 2015-04-27:
One piece doesn't help...

•git - used for version control 
•make, gcc-mingw, and g++ - used to build arduino.exe (this will also pull in gcc-core) 
•perl - use this version, activestate or other distros have trouble 
•unzip, zip - for dealing with archives 
When looking at the list, only perl showed up as a package.  I searched for others and got results that didn't make sense.  I am not a unix user and I suspect there is something being assumed here.

Makuna, 2015-04-27:
C:\GitHub\Arduino\build\build.xml:723: Execute failed: java.io.IOException: Cannot run program "unzip": CreateProcess er
ror=2, The system cannot find the file specified

Makuna, 2015-04-27:
It seems that Cygwin didn't add its bin to the path, so the build couldn't find unzip.  Adding C:\cygwin64\bin to the path fixed the problem above.

Makuna, 2015-04-28:
while the fixes seem to compile, I can't seem to get anything out of the uploaded code.