sandeepmistry/esp8266-Arduino #11: 32 bit or 64 ?

tytower, 2015-04-30:
I did not see what this library is for 
Is it 32 bit  64 bit  or both please?
My eyes have trouble with these greens and white pastel colours . Not good for older eyes.

tytower, 2015-05-01:
This from Sandeep 
"The toolchain is definitely 64-bit, so won't work on a 32-bit system. I'll
update the read me with that info this weekend.

Your best bet is to ask on https://github.com/esp8266/Arduino/issues/124 how
to build a 32-bit toolchain, then replace the 64-bit one with it."

sandeepmistry, 2015-05-02:
Question answered, closing ...

closed 2015-05-02 by sandeepmistry

tytower, 2015-05-09:
I just wanted to add Sandeep that The Arduino IDE can be had in 32 bit and 64 bit. It turns out my machine is in fact 64 bits but was running the 32 bit version because my Operating system "Puppy  Precis 5.7.1 Linux" is only available in 32bit.  So no doubt there will be some more who drop through the cracks and need a 32 bit version . I compiled some files for Igrr on the source for this fork and they would have been done in 32 bit so I am hoping this will not take long now to come to fruition.