sandeepmistry/esp8266-Arduino #12: OSC library testing/platform.txt modification

brendanmatkin, 2015-05-02:
Not sure if this is the right place to post this.. 

I've modified the OSC library to work on ESP8266 using v0.0.3! Is there anyone that can try it out? https://github.com/brendanmatkin/OSC 

One ESP issue was that this required a modification of the platform.txt file in Arduino\hardware\esp8266com\esp8266. I had to add -Wconversion-null flag to compiler.c.flags to hide several 'converting to non-pointer type ____ from null' warnings. No other board throws these warnings and it seems like the null returns are required by the library. Is this something that should be changed here, fixed in the OSC library (lines 116 to 197 in OSCMessage.cpp), or something else?

Once the OSC library is working here, I'm guessing it probably should be merged into the the official OSC branch or included upstream in this distribution? If it's only used for ESP then the library could be significantly downsized (right now it's very large: it includes SLIP encoding and AVR timings that could be left out) I'm a little new to actually fixing other repos so correct me if I'm messing up protocol! 

Thanks!

sandeepmistry, 2015-05-02:
@brendanmatkin It's a bit strange the library is returning NULL values for functions that don't return a pointer, for example:

```c
int32_t OSCMessage::getInt(int position){
	OSCData * datum = getOSCData(position);
	if (!hasError()){
		return datum->getInt();
    } else {
        return NULL;
    }
}
```
Maybe 0 should be returned instead? That might be better than changing ```platforms.txt```

In terms of changes to ```platform.txt```, I'm trying to keep this fork as close to [upstream](https://github.com/esp8266/Arduino) but in Arduino add-on format. So, it's probably best to submit a pull request in [esp8266/Arduino](https://github.com/esp8266/Arduino) and the changes will trickle down when I sync.

brendanmatkin, 2015-05-03:
I hope you don't mind me asking a follow up here for the OSC library:

It is possible that many of those methods could return 0 as an actual value, which I imagine is why they used null. Is there another option that you know of that would be a valid non-number, non-pointer return?

sandeepmistry, 2015-05-03:
@brendanmatkin probably, even -1 might be an acceptable return value.

Probably best to leave it as is for now (it's only a compiler warning), and discuss with the OSC author's what their goal was for handling errors.

commit 260aee75 (2015-03-30): Fix cbuf::read and cbuf::peak return types

Fixes #12
closed 2015-05-03 by sandeepmistry