sandeepmistry/esp8266-Arduino #14: MSG- Error Compiling  --- incorrect generation of g++.exe pathname

tawpgk, 2015-05-06:
IDE generates:
C:\Users\Tom\AppData\Roaming\Arduino15\packages\esp8266\tools\esptool\0.4.2/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++

File location:

C:\Users\Tom\AppData\Roaming\Arduino15\packages\esp8266\tools\esptool\0.4.2\xtensa-lx106-elf\xtensa-lx106-elf\bin\g++.exe

Please correct, I don't know where to look/edit to fix it myself.
TIA,
Tom

tawpgk, 2015-05-06:
Well I guess its my installation somehow, I have it fixed for the most part. My Arduino libraries and ..sketchbook are not on the C: drive and also I didnt copy over the ESP8226com directory...it finds the boards.txt file and listss the 8226, but not the esptool. 

Any clues are appreciated.

TIA

closed 2015-05-06 by tawpgk

sandeepmistry, 2015-05-06:
@tawpgk no need to select esptool, just hit the upload button!

See: https://github.com/esp8266/Arduino/pull/25