sandeepmistry/esp8266-Arduino #15: VisualMicro can't find the esptool

RussMathis, 2015-05-06:
using:
Visual Studio 2013 Professional
Arduino 1.6.3
this zip, stored to the hardware folder in the sketch directory
VisualMicro latest version as of(5/5/2013) for Arduino 1.6.3

Project compiles and uploads OK from Arduino 1.63
However, when using VisualMicro, after compile/build there's an upload error:

Uploading to I/O board using 'esptool'
Uploader started for board Generic ESP8266 Module
Uploader will use programmer name: esptool
Error - Uploader was unable to find the required programmer: esptool on platform esp8266
Upload failed. Did you select the correct upload method for this board?
System.Collections.Generic.KeyNotFoundException: The given key was not present in the dictionary.
   at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at processing.app.debug.AvrdudeUploader.uploadUsingPreferences(String buildPath, String className, Boolean verbose, Boolean ProgrammerUploadUsing, String mcuProgrammerOverrideName)
   at processing.app.SketchEditor.Upload(Boolean verbose)

Can anybody help with this? Guessing something missing the the platform.txt file?

sandeepmistry, 2015-05-06:
@RussMathis unfortunately I don't have access to a Windows machine to try this out.

Maybe @luc-github @rogerclarkmelbourne @brendanmatkin can help you out ...

The upload tool is set in boards.txt: https://github.com/sandeepmistry/esp8266-Arduino/blob/master/esp8266com/esp8266/boards.txt#L7 - maybe VisualMicro doesn't respect it.

rogerclarkmelbourne, 2015-05-06:
(lol) just installed VS2013 + the addon this morning, for my STM32 repo see http://stm32duino.com/viewtopic.php?f=20&t=75

However I've not had time to test it yet even with my STM32 repo

@sky reported needing to change platform.txt due pathing issues, see that post I linked to

However I'm not sure he installed my repo in the right place

I will take a look and report back when I get chance

skyng22003, 2015-05-07:
Hmm, I have mine installed under Arduino_1.6.0\hardware\Arduino_STM32, but everything works fine in the Arduino IDE is just when you use Visual Micro that it doesn't seem to be able to resolve the {runtime.hardware.path}

rogerclarkmelbourne, 2015-05-07:
See my post on the forum, the issue is that Visual Micro doesn't support the new plaftform.rewrite.keys feature that was added in 1.6.3, so that legacy hardware (anything prior to 1.6.2) would work with the new storage location of the ARM, Intel and other third party compilers (installed by the IDE)

If you look in platform.rewrite.keys it contains
```
old.0.compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/
new.0.compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
```

Basically its just a fancy text replace system that the IDE team had to implement because they hadnt considered the ramifications of moving the compiler binaries to a new location in 1.6.2

However if you see the forum there are other issues with Visual Micro :-(

It doesn't support the sub menus that allow the selection of processor variant for the Generic STM32F103C series,   or the F103R or F103Z series (and V series will be added soon) 

It also doesn't support the sub menu for upload method or the bootloader version (on Maple and Maple mini)

So as it stands its not usable.

Also. I'm not sure if it was VS2013 or Visual Micro, but I got a message about a "trial license" and also something ran the encryption system on my PC and generated an excryped file. I suspect this is also Visual Micro (though it could be VS2013), either way it was worrying to get a system alert about files being encrypted as I thought for a while I may have an encryption Trojan.
(well I hope it was Visual Micro installing a trial license, otherwise I could be in big problems)

sandeepmistry, 2015-05-07:
@rogerclarkmelbourne @skyng22003  thanks for looking into this and the info!

@RussMathis I'm marking this a "won't fix", VisualMicro isn't up to spec with the [Arduino IDE 3rd party Hardware specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification)

closed 2015-05-07 by sandeepmistry

rogerclarkmelbourne, 2015-05-08:
Yes. Its not a fault in your repo, its a shortcoming with Visual Micro.
I guess they may fix it, but either way is not an issue for you.