sandeepmistry/esp8266-Arduino #16: Links not found in Linux library

tytower, 2015-05-07:
Just playing around I noticed when extracted from the zip the links are not found . Perhaps the paths are wrong or name too lonh . See screenshot if I can get it here

tytower, 2015-05-07:
Screenshot on here for a week or less
http://www.filedropper.com/espsave1

tytower, 2015-05-07:
http://www.filedropper.com/espsave1_1

sandeepmistry, 2015-05-09:
@tytower is this causing a problem or just something you noticed? The links are relative so I don't think it is path length related..

P.S.: please attach screenshot to the issue directly next time, it's a bit awkward to get them from filedroppper.

tytower, 2015-05-09:
I am not sure if its "the problem" but the files they point to are not being found by the system . Thats why they show in red . They do seem to be there ,one directory up ,but just not the ones the links target. 

In relation to screenshots I found I could not attach a jpeg to git hub pages . I would much rather do that if its available thanks.

You may have noticed I posted on igrr 's site also to say there is something more to this 64 /32 bit issue . I made an installation on another machine  Using his latest and 1.6.4 Arduino all at 64 bits and got exactly the same error in compiling a sketch. 

That definately needs deeper investigation by one of you at least.

tytower, 2015-05-09:
Just had another go at attaching a png or jpeg -could be a problem with my browser

tytower, 2015-05-10:
I got your latest push and tried it on my latest arduino IDE 1.6.4 and got the same error report 

Arduino: 1.6.4 (Linux), Board: "Generic ESP8266 Module, 80 MHz, 57600, 512K"

Cannot run program "/opt/arduino-1.6.4-linux64/arduino-1.6.4/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" (in directory "."): error=2, No such file or directory
I'm gonna see if I can get an early IDE and try that

tytower, 2015-05-10:
I copied esp8266.com into hardware as you said. Just didn’t want it in my sketchbook . The error  directory is different to the files location , see top path compared to below

Cannot run program “/opt/arduino-1.6.4-linux64/arduino-1.6.4/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++” (in directory “.”): error=2, No such file or directory

  
Its looking in the wrong directory!
hardware/tools 
not 
hardware/esp8266com/esp8266/tools

So how do I fix that?
[image]

tytower, 2015-05-10:
happy to say i seem to have advanced to be able to compile in 1.6.4 your version 5 but had to move the libraries around a bit and have duplicate libraries atm but at least it works on 64 bit now

tytower, 2015-05-10:
I will try it when I get time on 32 bit. The problem is not what igrr says it is.

sandeepmistry, 2015-05-10:
@tytower can we close this issue? Your original concern on "links not found" doesn't seem to be causing a real problem.

I'm still confused about the error message in your screenshot, the path should be ```esp8266com/esp8266/tools/linux/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++``` - I don't see ```linux``` in the error message. This is configured by: https://github.com/sandeepmistry/esp8266-Arduino/blob/master/esp8266com/esp8266/platform.txt#L11

Which libraries did you have to move around?

tytower, 2015-05-10:
So many tries I lost count . Essentially the IDE looks where it isn't so the library has to be moved where it finds it . Simplistic I know but that worked on 64bit

sandeepmistry, 2015-05-11:
Closing, as "links not found" doesn't seem to be an issue ...

closed 2015-05-11 by sandeepmistry