sandeepmistry/esp8266-Arduino #17: WiFi.begin doesn't work

JohnSL, 2015-05-09:
I haven't been able to get WiFi.begin to work with version 0.0.4 along with IDE 1.6.3 downloaded from Auduino.cc. Here is the code that gives me problems:

`WiFi.begin(ssid, password);`

This is the output that I see from the Serial monitor as a result of this call:

```
sl
scandone
usl
sul 7 0
Status: 6reconnect
rm match
pm close 7 0 0/14711153
scandone
add 0
aid 5
pm open phy_2,type:2 0 0

connected with socha, channel 1
dhcp client start...
cnt 
```
Note: when I use IDE 1.6.1 downloaded from the esp8266/Arduino GitHub repository, the code above works perfectly without sending any output to the Serial monitor.

Is this library compatible with IDE 1.6.3? If not, what version of the IDE should I be using?

sandeepmistry, 2015-05-09:
@JohnSL IDE 1.6.3 should be fine. I just tested [ESP8266WebServer Hello Server example](https://github.com/sandeepmistry/esp8266-Arduino/blob/master/esp8266com/esp8266/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino) on my Mac with master, no problems with setting up wifi.

I've pushed a new 0.0.5 release for you to try out.

JohnSL, 2015-05-10:
Thanks, 0.0.5 fixed that problem. I verified it using the NTPClient sample. I then went back to 0.0.4 and tried the same sample and verified that 0.0.4 does not work.

closed 2015-05-10 by JohnSL