sandeepmistry/esp8266-Arduino #20: No function for stopping AP Server

dmason1992, 2015-06-30:
Have been unable to stop and AP server once it is started (despite disconnecting and unpowered devices, reflashing other scripts etc.)

below is arduino script I used

#include <ESP8266WiFi.h>
#include <WiFiUDP.h>

void setup() {
  // put your setup code here, to run once:
  WiFi.softAP("openWiFi");

}

void loop() {
  // put your main code here, to run repeatedly:

}

dmason1992, 2015-06-30:
Ok I found out this can be solved by using ESP.reset() - Might be work adding a short note in the documentation that this is all it takes :)

dmason1992, 2015-06-30:
Setting WiFi mode to: WiFi.mode(WIFI_AP); or WiFi.mode(WIFI_OFF); stops the access point from being visible but when you try to create a new access point with a different name etc. it simply creates the same one you had before

dmason1992, 2015-07-01:
Further Update it would appear that it is actually the setting up a secure AP that is not working at all.

sandeepmistry, 2015-07-05:
@dmason1992 I'm not really maintaining this fork/repo anymore as the upstream repo now supports install via the board manager.

I suggest try the latest from https://github.com/esp8266/Arduino and if you still have problems create an issue there.

shirish47, 2015-08-07:
I want to Create an AP to host a small page to get a network ID and Password for the router. Based on problem mentioned above .. I think its not possible. is it so? how to stop ap and change the mode??